PrusaOwners/prusaowners #4: Please delete the content for which I am the sole contributor

matthew-humphrey, 2018-08-28:
Please delete the following content, for which I am the sole contributor:

Bed_Leveling_with_Wave_Springs.md
Cold_Pull_on_MK3.md
Moons_stepper_motors.md
PINDA_Temperature_Calibration_with_Electrical_Continuity.md
images/Accessing_the_filament.png
images/Cleaning_Filament.JPG
images/Filament_Sensor_Light_Block.png
images/HoldingSleeveSpringAndSpacer.JPG
images/Idler_Cover_Screws.jpg
images/MoonsWiring.png
images/Moons_Extruder.JPG
images/Octoprint-MeshMap-Screenshot.PNG
images/PINDA_cal_with_continuity_measurement_setup.jpg
images/PINDA_cal_with_continuity_results.jpg
images/Skelestruder_Legend.jpg
images/Skelestruder_Legend_Alt_View.jpg
images/Sleeve.JPG
images/SleeveRetainingSpring.JPG
images/SpringInstalled.JPG

VileBrigandier, 2018-08-28:
Done.

closed 2018-08-28 by VileBrigandier

VileBrigandier, 2018-08-28:
Contributions cleared from git history as well.