PrusaOwners/prusaowners #22: Suggestion for wave spring installation page

jcwren, 2018-09-03:
First, it looks like two of the pictures are inverted (washer with straw on table, pliers holding straw). No big deal, the idea is still conveyed. 

I haven't yet installed my wave-spring washers, but instead of the straw and pliers dance, what about using toothpicks? Put the toothpick in the screw hole, drop the spacer and washer on to it, then lower the print bed onto the toothpicks, one by one replace each toothpick with a screw. 

It might be necessary to put a piece of tape on the bottom of the hole, and possibly clip the toothpick to a shorter length to make aligning the bed on top of them easier.

I plan to give this a shot when the wave-spring washers I ordered are delivered.

VileBrigandier, 2018-09-03:
Hi jcwren,

1. The flipped images are due to the image submitter's device using EXIF data to determine image rotation, and that breaks on some browsers. Re-opened issue #1 regarding this. The issue was closed due to the images being removed for other reasons, but now that they are back the issue is valid again.
2. The improvements you are proposing may be valid improvements for the method; however, there's also a method for doing this in a much more rigid way: https://github.com/PrusaOwners/prusaowners/wiki/Bed_Leveling_without_Wave_Springs - There may be pros/cons for each, I haven't verified the new method myself, just wanted to point this out before you tear into this using the previous method.

sarf2k4, 2018-09-08:
@VileBrigandier, I few questions regarding bed leveling without wave springs (sorry for being off topic). 

I read the guide thoroughly, during the print temperature calibration, in the guide stated for example 60'c is my usual print temperature and I would have to level the bed manually with 60'c.

1. What's not mentioned is that how was the bed elevation results if the bed temperature differs from 60'c for example 80 or 100'c?
2. What about the temperature calibration, should it be turned off?
3. Do I have to run temperature calibration after doing the locknut bed leveling method?

I am planning to print several types of materials, primarily abs and pla, so this is a concern to me actually.

Thank you in advance

VileBrigandier, 2018-09-08:
@sarf2k4,

Leefla in the Prusa Discord is the author of that article and should be able to answer that. I have sold my MK3 and moved the wiki content here from a MediaWiki installation, and unfortunately it set my name on a lot of the articles. Hopefully that will fix itself with time.

Thanks!