PrusaOwners/prusaowners #54: Skelestruder patch for indirect filament sensor appears incomplete

tg73, 2019-01-05:
@jltx1 From what I can tell, there are two functions that need patching: `pat9125_update()` and `pat9125_update_y()`. There is also `pat9125_update_y2()`, but this does not appear to be used anywhere. All three functions contain the expression `pat9125_y -= iDY`. The [patch ](https://github.com/PrusaOwners/prusaowners/blob/master/Skelestruder_Assembly.md#editing-needed-below) only changes `pat9125_update_y()`. Is this correct?

jltx1, 2019-01-05:
When I made that patch, the only function being used for filament sensing was the one, but they may have expanded it in recent fw updates.  I currently flip all three just in case.

AbeFM, 2019-01-09:
I found two to change, and I think I ended up changing just one. and it's
working ok with ZB firmware.

line 19 pat9125.c
      pat9125_y -= iDY; //negative number, because direction switching does
not work//Abe Skelestruder Original
//      pat9125_y += iDY; //negative number, because direction switching
does not work//Abe Skelestruder Added

I didn't make the change, works fine not done.

Line 216
//Abe Skelestruder      pat9125_y -= iDY; //negative number, because
direction switching does not work
      pat9125_y += iDY; //negative number, because direction switching does
not work //Abe Skelestruder

I flipped this one.

Line 324:
     pat9125_y -= dy; //negative number, because direction switching does
not work
//      pat9125_y += dy; //negative number, because direction switching
does not work //Abe Skelestruder
Looks like I kept that one stock as well.

From what I'm running right now, it seems you only need to change the one
line ~216.