PrusaOwners/prusaowners #57: Where is the prusa config file?

JBTERRA, 2019-02-03:
The guide speak about a mk3 config file but I can't find one to download. I have a MK2s but I hope to get this file at a starting point to configure klipper

Arksine, 2019-02-03:
It is in the PrusaOwners repo.  I have plans to merge it upstream, there are other things that need to be handled first.

https://raw.githubusercontent.com/PrusaOwners/klipper/master/config/printer-prusa-i3-mk3-2017.cfg

JBTERRA, 2019-02-04:
Thank you. Btw I've got a prusa MK2 and I've tried to adapt the config with the generic mini rambo example. It kinda work but  can't see the speed processing I thought, it seems like before with a gcode I'm using for testing. 

I'm using the mk2 also as a laser engraver. With original prusa firmware (marlin) I can scan grayscale images buy it seems I hit the limit of minirambo processing capacity for gcode so I've got a cap in speed.

Thought that going to klipper could give me more speed in scanning but it seems the same.

Maybe I've one something wrong or Is normal that I can't fast scan even with klipper? (maybe is more oriented to speed up prints but gcode processing speed is the same)

Hope you can give me some hints, thanks