PrusaOwners/prusaowners #60: Einsy + Banan Pi Zero + Klipper + DWC2

rafaljot, 2019-04-18:
I'm very disappointed with Rassberry Pi Zero and OctoPrint. So I made mix of
- Banan Pi Zero ,  the same size and compatible with rPi Zero, Allwiner H2+ with 4 cores, external antenna http://www.banana-pi.org/bpi-zero.html
- Ubuntu Base Minimal by @avafinger https://github.com/avafinger/bananapi-zero-ubuntu-base-minimal
- Klipper,
- DWC2 for Klipper https://github.com/Stephan3/dwc2-for-klipper

It works for now, Requires minor corrections. Boot takes ~15s. I think I can tune it and cut ~3s.
I think it is good idea to make full SD image with this setup.

What do you think about this setup ?

brunnels, 2020-01-08:
I'm running a similar setup to you and have some questions:

1.  what 0.9 degree steppers are you using and what TMC2130 settings are you using?  I think I got too high amperage steppers so I had to add heatsinks to the bottom of the einsy and a fan otherwise the extruder would skip and jam or I would get layer shifting.

2.  what do the DLP printed pulleys buy you over standard aluminum pulleys?

rafaljot, 2020-01-12:
@brunnels 
1. I have 1.7amp 09 motors + fan + heatsinks on both sides.
2. I use resin pulleys just for fun :)

tillin9, 2021-02-11:
Pleasantly surprised to see this - I run a very similar setup with the exception I haven't made the jump to Klipper - though thinking about it, Did you get any improvement in print quality or speed? 

I'm still not 100% satisfied with my Prusa's quality - I occasionally have VFAs depending on settings/ part geometry. Faster is always better. Was looking at replacing the Einsy with an SKR or going to Klipper to see if that helped. 

Is the print quality better for you? 

P.S. I love the Banana Pi Zero. Designed this - https://www.prusaprinters.org/prints/32433-einsy-enclosure-for-bpi-banana-pi-m2-zero-plus (my printer hadn't been upgraded to a bear at that time)