PrusaOwners/prusaowners #61: Power Cable Update (PSU to Einsy)

Gefionious, 2019-06-04:
The parts for the power cables are incorrect for the mk3 PSU to Einsy. I was able to get the drawings from Prusa. They definitely have different spade connector sizes at either end, where your docs are incorrect. Also, there are some great PSU mounting designs on thingiverse for the Meanwell 320W PSU, which I love and which I need slightly longer cables for.
[image]

elektronchika, 2019-07-03:
Is this the drawing for the 240W or 320W power supply? If this is for the 240W, how long shall the cables be for the 320W one?

How long are the cables for the heated bed?

Thanks!!!

Gefionious, 2019-07-03:
This drawing is from Prusa for the standard i3 mk3/s 240W PSU with the recommended cable management. You would need it to be approximately 1"/25mm longer for the Meanwell 320W PSU using this mount from thingiverse [https://www.thingiverse.com/thing:3082178]. I have done no work on the heated bed, but they would be independent and required no changes on my end with the new PSU (i.e. the heated bed is not directly attached to the PSU, it is from the Einsy board)

Gefionious, 2019-07-05:
This drawing is for the 240W standard PSU, they would need to be about an inch or 25mm longer for the meanwell 320W PSU mounted with one of the thingiverse mounts: https://www.thingiverse.com/thing:3082178

mkolassa, 2022-08-08:
For those who come after...

So far I've tracked down two sets of flanged fork connectors which may be the correct items.

The first comes from a-dude16 on the Prusa forums at...

https://forum.prusa3d.com/forum/original-prusa-i3-mk3s-mk3-user-mods-octoprint-enclosures-nozzles/extending-the-psu-cables./paged/3/#post-227137

> Not sure if this helps anyone.
> KSS
> PSU side is LYF2-4
> Einsy board side is LYF1-3G

The other ones are from Morris Products:

PSU Side: Morris Products 11810 Nylon Insulated Flange Fork/Spade Terminals - 16-14 Wire, # 6 Stud
Einsy Side: Morris Products 11802 Nylon Insulated Flange Fork/Spade Terminals - 22-16 Wire, # 4 Stud

mkolassa, 2022-08-08:
Looks like there's a discrepancy in the spec sheets provided by Morris Products.  It may actually be the 11812 and 11804.

I'm ordering both sets and will confirm the correct one once I get them.

mkolassa, 2022-08-26:
Sadly, my orders had to be cancelled as the vendor was actually out of stock.

It's incredibly difficult to find a vendor that doesn't require purchase of a bag of 100x of these things, at least in the USA.

jhoblitt, 2022-08-26:
Terminals can now be purchased directly from te.com in "sample" quantities (single units). Tyco is superior molex iny experience.