PrusaOwners/prusaowners #65: Klipper on MK3S

brotherdust, 2019-12-03:
I followed the guide to getting Klipper set up on MK3S, but ran into an issue on the firmware flash step. The printer gets stuck on "updating... do not disconnect". The flash utility indicates that the flash was successful.

Rebooted the printer and it's just stuck at the boot screen now (Prusa i3...). Any thoughts?

Arksine, 2019-12-03:
Most likely its an issue with the configuration, there have been some config breaking changes recently and I haven't had a chance to update the sample configuration.  If you check /tmp/klippy.log it will tell you what is wrong.

brotherdust, 2019-12-03:
Here's the result of the flashing process:
```
pi@octopi:~/klipper $ make flash FLASH_DEVICE=/dev/serial/by-id/usb-Prusa_Research__prusa3d.com__Original_Prusa_i3_MK3_CZPX1519X004XK20483-if00
  Flashing out/klipper.elf.hex to /dev/serial/by-id/usb-Prusa_Research__prusa3d.com__Original_Prusa_i3_MK3_CZPX1519X004XK20483-if00 via avrdude

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.01s

avrdude: Device signature = 0x1e9801 (probably m2560)
avrdude: reading input file "out/klipper.elf.hex"
avrdude: writing flash (23720 bytes):

Writing | ################################################## | 100% 4.19s

avrdude: 23720 bytes of flash written
avrdude: verifying flash memory against out/klipper.elf.hex:
avrdude: load data flash data from input file out/klipper.elf.hex:
avrdude: input file out/klipper.elf.hex contains 23720 bytes
avrdude: reading on-chip flash data:

Reading | ################################################## | 100% 3.05s

avrdude: verifying ...
avrdude: 23720 bytes of flash verified

avrdude: safemode: Fuses OK (E:FD, H:D0, L:FF)

avrdude done.  Thank you.
```

Here's the output from `/tmp/klippy.log`:
```
cat /tmp/klippy.log
Starting Klippy...
Args: ['/home/pi/klipper/klippy/klippy.py', '/home/pi/printer.cfg', '-l', '/tmp/klippy.log']
Git version: 'v0.6.0-1892-g5aa5b0f4'
CPU: 4 core ARMv7 Processor rev 4 (v7l)
Python: '2.7.13 (default, Sep 26 2018, 18:42:22) \n[GCC 6.3.0 20170516]'
Start printer at Tue Dec  3 12:36:45 2019 (1575401805.1 9.4)
===== Config file =====
[probe]
pin = PB4
x_offset = 24
y_offset = 5
z_offset = 0
speed = 10.0

[bed_mesh]
speed = 100
min_point = 11,1
max_point = 215,193
probe_count = 3,3
mesh_pps = 2,2
algorithm = lagrange

[homing_override]
gcode =
        G1 Z3
        G28 X0 Y0
        G1 X101 Y100 F5000
        G28 Z0
axes = Z
set_position_x = 0
set_position_y = 0
set_position_z = 0

[gcode_macro G80]
gcode =
        G28
[... 232 lines cut ...]
3: (215.0, 97.0)
4: (113.0, 97.0)
5: (11.0, 97.0)
6: (11.0, 193.0)
7: (113.0, 193.0)
8: (215.0, 193.0)
Declaration of '__filament __load' hides previous menuitem declaration
Declaration of '__filament __unload' hides previous menuitem declaration
Declaration of '__filament __feed' hides previous menuitem declaration
Extruder max_extrude_ratio=20.787584
Config error
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/klippy.py", line 138, in _connect
    self._read_config()
  File "/home/pi/klipper/klippy/klippy.py", line 135, in _read_config
    pconfig.check_unused_options(config)
  File "/home/pi/klipper/klippy/configfile.py", line 260, in check_unused_options
    section,))
Error: Section 'prusa_gcodes' is not a valid config section
```

Here's the result on the printer screen:
[image]

Hopefully that helps?

brotherdust, 2019-12-03:
Update: taking the `[prusa_gcodes]` section out of the printer.cfg fixed it. =)

closed 2019-12-03 by brotherdust